jetperch/pyjoulescope #1: Consider using context managers for scan operations

markcampanelli, 2019-07-21:
Consider using a Pythonic [context manager](https://docs.python.org/3/library/contextlib.html#contextlib.contextmanager) for scan operations.

Using the context manager pattern,
```python
js = joulescope.scan_require_one()
js.open()
try:
    js.parameter_set('source', 'on')
    js.parameter_set('i_range', 'auto')
    data = js.read(contiguous_duration=0.25)
finally:
    js.close()
```
becomes the more streamlined
```python
with joulescope.scan_require_one() as js:
        js.parameter_set('source', 'on')
        js.parameter_set('i_range', 'auto')
        data = js.read(contiguous_duration=0.25)
```
where the context manager handles closing the device, doing so even when an exception is raised.

mliberty1, 2019-07-22:
I took a quick look at this, and adding a context manager is straightforward.  Adding a config is also easy, but I would want it to take a config string with a mode rather than detailed parameter setttings.  The parameters can vary from device to device (they don't currently), so the driver should abstract the details, at least at this level.  How about:

    :param config: The initial default configuration following device open.
        Choices are ['auto', 'off', 'ignore', None].
        * 'auto': enable the sensor and start collecting data with
          current sensor autoranging.
        * 'ignore' or None: Leave the device in its existing state.
        * 'off': Turn the sensor off and disable data collection.

Then the use case becomes:

    import joulescope
    with joulescope.scan_require_one(config='auto') as js:
        data = js.read(contiguous_duration=0.25)

I went ahead an implemented this in commit [fa8f3ded](https://github.com/jetperch/pyjoulescope/commit/fa8f3ded120c2c5ee212a1e85656d8e5654e1fc1).  Let me know if you have feedback!

mliberty1, 2019-07-31:
Available in 0.5.0.

closed 2019-07-31 by mliberty1