jetperch/pyjoulescope #12: Simple example failing to open device on macOS

petertorelli, 2020-06-11:
Hello,

The simple example from `pyjoulescope_examples` is working fine on Ubuntu 18.04 (x86) and Win10, but is now failing on macOS (catalina 10.15.4), it used to work in the past. (The only thing that changed is I've moved to catalina recently from Mojave.)

What additional debug info I can collect for you?

```
% python3
Python 3.7.5 (default, Nov  1 2019, 02:16:32) 
[Clang 11.0.0 (clang-1100.0.33.8)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import joulescope as js
>>> print(js.__version__)
0.8.14
>>> 
```

Failure:

```
26 ✗ [11:52:31 Thu Jun 11] ~/eembc/git/pyjoulescope_examples/bin
% python3 ./capture_simple.py
libusb_set_configuration 1 failed [-4]
open failed: 16d0/0e88/000722
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/joulescope/usb/libusb/device.py", line 790, in open
    may_raise_ioerror(rv, 'libusb_set_configuration 1 failed')
  File "/usr/local/lib/python3.7/site-packages/joulescope/usb/libusb/device.py", line 732, in may_raise_ioerror
    raise IOError(s)
OSError: libusb_set_configuration 1 failed [-4]
While running command
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/joulescope/usb/device_thread.py", line 64, in _cmd_process
    rv = self._device.open(event_callback_fn)
  File "/usr/local/lib/python3.7/site-packages/joulescope/usb/libusb/device.py", line 790, in open
    may_raise_ioerror(rv, 'libusb_set_configuration 1 failed')
  File "/usr/local/lib/python3.7/site-packages/joulescope/usb/libusb/device.py", line 732, in may_raise_ioerror
    raise IOError(s)
OSError: libusb_set_configuration 1 failed [-4]
usb control transfer failed 5
calibration_read transfer failed 5
could not set defaults
Device.stop() while attempting _stream_settings_send
Traceback (most recent call last):
  File "./capture_simple.py", line 33, in <module>
    run()
  File "./capture_simple.py", line 26, in run
    data = device.read(contiguous_duration=0.001)
  File "/usr/local/lib/python3.7/site-packages/joulescope/driver.py", line 797, in read
    self.start(on_stop, duration=duration, contiguous_duration=contiguous_duration)
  File "/usr/local/lib/python3.7/site-packages/joulescope/driver.py", line 725, in start
    self._stream_settings_send()
  File "/usr/local/lib/python3.7/site-packages/joulescope/driver.py", line 585, in _stream_settings_send
    _ioerror_on_bad_result(rv)
  File "/usr/local/lib/python3.7/site-packages/joulescope/driver.py", line 48, in _ioerror_on_bad_result
    raise IOError('usb.Device %s' % (rv,))
OSError: usb.Device ControlTransferResponse(result=5)
```

mliberty1, 2020-06-11:
That's weird.  I just tried the same on my MacBook with 10.15.5, and it worked great.  Perhaps the update to Catana messed up libusb.  Is everything up to date?  How about trying this:

    brew update
    brew reinstall libusb
    pip3 uninstall joulescope
    pip3 install -U joulescope
    python3 ./capture_simple.py

Does that change anything?

petertorelli, 2020-06-11:
Yes, it works now. I wish I knew what was wrong with the brew libusb, but that was the problem because I had tried reinstalling joulescope before filing the issue but that did nothing. It's always a tiny bit unnerving when a random reinstall of a key driver fixes things. :)

Thanks.

closed 2020-06-11 by petertorelli

mliberty1, 2020-06-11:
Yeah, I agree that it's a bit unnerving.  In this case, Catalina requires a new libusb bottle.  I am not sure how brew keeps in sync with macOS updates, but it obviously did not.  

For what its worth, here's the metadata:
https://formulae.brew.sh/api/formula/libusb.json

And here is how the Joulescope UI manages packaging the right versions:
https://github.com/jetperch/pyjoulescope_ui/blob/master/joulescope_ui/libusb_mac.py