jetperch/pyjoulescope #19: Windows 10 Running as admin / non admin yeild different values

efox29, 2020-12-09:
Upgraded to 0.9.3 from 0.9.2 (dirty update - no uninstall). Installer installed in C:\Program Files (x86)\Joulescope and required admin rights on my work computer to install. 

When using, my values were around 300-700uA when they should be < 200uA. 

Downgraded to 0.9.2 (dirty downgrade). Same behavior.

Rebooted machine. Same behaviour.

Uninstalled 0.9.2 and installed 0.9.3. Same behaviour.

Ran 0.9.3 as admin, and the values are < 200uA. 

No admin
[image]

Admin rights
[image]

mliberty1, 2020-12-09:
HI @efox29 - This issue is certainly strange.  Judging by the noise levels, I am guessing that the current range (or something else) is different between the two collections.  Could you check that the current ranging is set the same in both cases?  I normally recommend "auto", but just make sure that they are the same.  

Your Joulescope stores config in %LOCALAPPDATA%, so perhaps this is different between the two runs.  I just tried running as admin on my machine, and it seems to pull the same configuration settings.  What do you see?

efox29, 2020-12-09:
@mliberty1 There are 2 configs one for my domain user account and one for the elevated local user (in their respective %localappdata% directories.

I thought they were both set to auto...

[image]

Made it auto, and now I am able to get the correct values.

closed 2020-12-09 by efox29

mliberty1, 2020-12-09:
Great!  I'm glad that this was something easy...