jetperch/pyjoulescope #24: Unable to install joulescope via Pipenv

digitalorder, 2021-11-25:
`joulescope` can safely be installed directly on my machine:

```
test-joulescope $ pip3 install joulescope
Collecting joulescope
  Downloading joulescope-0.9.7.tar.gz (602 kB)
     |████████████████████████████████| 602 kB 4.1 MB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting pymonocypher>=0.1.3
  Downloading pymonocypher-3.1.0.0.tar.gz (100 kB)
     |████████████████████████████████| 100 kB 11.5 MB/s
  Preparing metadata (setup.py) ... done
Requirement already satisfied: python-dateutil>=2.7.3 in /Users/digitalorder/.asdf/installs/python/3.8.9/lib/python3.8/site-packages (from joulescope) (2.8.2)
Collecting numpy>=1.17
  Using cached numpy-1.21.4-cp38-cp38-macosx_10_9_x86_64.whl (16.9 MB)
Requirement already satisfied: psutil in /Users/digitalorder/.asdf/installs/python/3.8.9/lib/python3.8/site-packages (from joulescope) (5.8.0)
Collecting pyjls>=0.3.3
  Downloading pyjls-0.3.4.tar.gz (243 kB)
     |████████████████████████████████| 243 kB 10.6 MB/s
  Preparing metadata (setup.py) ... done
Requirement already satisfied: six>=1.5 in /Users/digitalorder/.asdf/installs/python/3.8.9/lib/python3.8/site-packages (from python-dateutil>=2.7.3->joulescope) (1.16.0)
Building wheels for collected packages: joulescope, pyjls, pymonocypher
  Building wheel for joulescope (pyproject.toml) ... done
  Created wheel for joulescope: filename=joulescope-0.9.7-cp38-cp38-macosx_11_0_x86_64.whl size=512786 sha256=2d27138ac2e042e4d9e0bb0ef177be797e10f47431f845356c0bbec0a1b8c424
  Stored in directory: /Users/digitalorder/Library/Caches/pip/wheels/17/4c/e1/8a97f0e2c0bba539ab4bcaa152ad156341655e43736f8dbb45
  Building wheel for pyjls (setup.py) ... done
  Created wheel for pyjls: filename=pyjls-0.3.4-cp38-cp38-macosx_11_0_x86_64.whl size=173232 sha256=64020c272310d8b84c1b2f8d965027f218ec34889351b10d4c816b45e35e584f
  Stored in directory: /Users/digitalorder/Library/Caches/pip/wheels/12/aa/a4/356fbb8a81119d7c143049645e8f8d532a2f086a86620d62d2
  Building wheel for pymonocypher (setup.py) ... done
  Created wheel for pymonocypher: filename=pymonocypher-3.1.0.0-cp38-cp38-macosx_11_0_x86_64.whl size=91267 sha256=d8650d5b2bb16cfb0e52731a21b2bff25ce4a313a4336a6c0b4f934910978378
  Stored in directory: /Users/digitalorder/Library/Caches/pip/wheels/fa/dc/91/c095ac73b6a6a29bea175954b2c9348da64934bc18cb4d1ef1
Successfully built joulescope pyjls pymonocypher
Installing collected packages: numpy, pymonocypher, pyjls, joulescope
Successfully installed joulescope-0.9.7 numpy-1.21.4 pyjls-0.3.4 pymonocypher-3.1.0.0
```

Then I try minimal `pipenv` setup. My `Pipfile`:

```toml
[[source]]
name = "pypi"
url = "https://pypi.org/simple"
verify_ssl = true

[dev-packages]

[packages]
joulescope = "*"

[requires]
python_version = "3.8"
```

And it fails for me:

```
$ pipenv install
Warning: the environment variable LANG is not set!
We recommend setting this in ~/.profile (or equivalent) for proper expected behavior.
Creating a virtualenv for this project…
Pipfile: /Users/digitalorder/dev/test-joulescope/Pipfile
Using /Users/digitalorder/.asdf/installs/python/3.8.9/bin/python3 (3.8.9) to create virtualenv…
⠼ Creating virtual environment...created virtual environment CPython3.8.9.final.0-64 in 253ms
  creator CPython3Posix(dest=/Users/digitalorder/.local/share/virtualenvs/test-joulescope-a23cxIFY, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/Users/digitalorder/Library/Application Support/virtualenv)
    added seed packages: pip==21.3.1, setuptools==58.3.0, wheel==0.37.0
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator

✔ Successfully created virtual environment!
Virtualenv location: /Users/digitalorder/.local/share/virtualenvs/test-joulescope-a23cxIFY
Pipfile.lock not found, creating…
Locking [dev-packages] dependencies…
Locking [packages] dependencies…
✔ Success!
Updated Pipfile.lock (851909)!
Installing dependencies from Pipfile.lock (851909)…
An error occurred while installing pywin32==302 --hash=sha256:2393c1a40dc4497fd6161b76801b8acd727c5610167762b7c3e9fd058ef4a6ab --hash=sha256:251b7a9367355ccd1a4cd69cd8dd24bd57b29ad83edb2957cfa30f7ed9941efa --hash=sha256:48dd4e348f1ee9538dd4440bf201ea8c110ea6d9f3a5010d79452e9fa80480d9 --hash=sha256:496df89f10c054c9285cc99f9d509e243f4e14ec8dfc6d78c9f0bf147a893ab1 --hash=sha256:543552e66936378bd2d673c5a0a3d9903dba0b0a87235ef0c584f058ceef5872 --hash=sha256:79cf7e6ddaaf1cd47a9e50cc74b5d770801a9db6594464137b1b86aa91edafcc --hash=sha256:af5aea18167a31efcacc9f98a2ca932c6b6a6d91ebe31f007509e293dea12580 --hash=sha256:d3761ab4e8c5c2dbc156e2c9ccf38dd51f936dc77e58deb940ffbc4b82a30528 --hash=sha256:e372e477d938a49266136bff78279ed14445e00718b6c75543334351bf535259 --hash=sha256:fe21c2fb332d03dac29de070f191bdbf14095167f8f2165fdc57db59b1ecc006! Will try again.
  🐍   ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 9/9 — 00:00:30
Installing initially failed dependencies…
[pipenv.exceptions.InstallError]:   File "/Users/digitalorder/Library/Python/3.8/lib/python/site-packages/pipenv/core.py", line 1863, in do_install
[pipenv.exceptions.InstallError]:       do_init(
[pipenv.exceptions.InstallError]:   File "/Users/digitalorder/Library/Python/3.8/lib/python/site-packages/pipenv/core.py", line 1246, in do_init
[pipenv.exceptions.InstallError]:       do_install_dependencies(
[pipenv.exceptions.InstallError]:   File "/Users/digitalorder/Library/Python/3.8/lib/python/site-packages/pipenv/core.py", line 858, in do_install_dependencies
[pipenv.exceptions.InstallError]:       batch_install(
[pipenv.exceptions.InstallError]:   File "/Users/digitalorder/Library/Python/3.8/lib/python/site-packages/pipenv/core.py", line 763, in batch_install
[pipenv.exceptions.InstallError]:       _cleanup_procs(procs, not blocking, failed_deps_queue, retry=retry)
[pipenv.exceptions.InstallError]:   File "/Users/digitalorder/Library/Python/3.8/lib/python/site-packages/pipenv/core.py", line 681, in _cleanup_procs
[pipenv.exceptions.InstallError]:       raise exceptions.InstallError(c.dep.name, extra=err_lines)
[pipenv.exceptions.InstallError]: []
[pipenv.exceptions.InstallError]: ['ERROR: Could not find a version that satisfies the requirement pywin32==302 (from versions: none)', 'ERROR: No matching distribution found for pywin32==302']
ERROR: ERROR: Package installation failed...
```

Setup: 
- macOS 11.4
- Python 3.8 via `asdf`
- pipenv 2021.11.23

So missing `pywin32` on macOS is understandable, but not exactly what I would expect to pop up here.

Note 1: same thing when I try to do `pipenv install` on an Ubuntu host

Note 2: `Pipfile.lock` appears at the of a failed `pipenv install` and I can just delete offending `pywin32` and `pypiwin32` from it directly. But I would like to get to the bottom of the problem.

mliberty1, 2021-11-25:
TIL [pipenv](https://github.com/pypa/pipenv) exists, and it looks to not fully support the requirements.txt format.  The pyjoulescope [requirements.txt](https://github.com/jetperch/pyjoulescope/blob/master/requirements.txt) has this line:

    pypiwin32>=223; sys_platform == 'win32'

Since pywin32 shows up in the Pipfile, it looks like pipenv ignores sys_platform.  Here's a [StackOverflow](https://stackoverflow.com/questions/57315096/pipenv-dependencies-of-platform-specific-packages-are-installed-unconditionally) question.

digitalorder, 2022-01-20:
Thanks for the tip. These lines in _Pipfile_ actually help to some extent:

```
pypiwin32 = {version = ">=223", sys_platform = "== 'win32'"}
pywin32 = {version = "*", sys_platform = "== 'win32'"}
```

This solution still ignores lines like this from you `setup.py`

https://github.com/jetperch/pyjoulescope/blob/ab9645a2774cf5d5355dee4c1e60a566419b0e00/setup.py#L179-L180

for my linux-based setup. `pipenv` for whatever reason uses windows rule for `numpy>=1.20` and it conflicts with another package that have which requires `numpy==1.19`.

Anyway, I believe this topic is not relevant to this repository anymore because we start to debug `pipenv` here.