jetperch/pyjoulescope #25: Cannot Import Joulescope Lib

e-chesoni, 2022-03-10:
Hi,
I am having an issue importing the joulescope lib. 

`>>> import joulescope`

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Elaine Chesoni\anaconda3\lib\site-packages\joulescope\__init__.py", line 15, in <module>
    from joulescope.driver import scan, scan_require_one, scan_for_changes, \
  File "C:\Users\Elaine Chesoni\anaconda3\lib\site-packages\joulescope\driver.py", line 20, in <module>
    from joulescope.stream_buffer import StreamBuffer, DownsamplingStreamBuffer
  File "joulescope\stream_buffer.pyx", line 1, in init joulescope.stream_buffer
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject
```

The installation seemed to go fine, and after researching the numpy error, I downgraded numpy to 1.20.0.

Not sure what else I can do?

Please let me know if I can offer any more information. I love this joulescope product and would like to incorporate it in a custom python application!

mliberty1, 2022-03-10:
Hi @e-chesoni 

That's odd.  Are you running 32-bit python?  The wheels on pypi are only 64-bit.  You should be able to run 32-bit, but you will need to build them yourself.  If you are not sure, try running this command, modified for however you start python:

    python -c "import sys; import platform; print(f'{platform.platform()}\n{sys.version}')

e-chesoni, 2022-03-10:
Hi @mliberty1 !
No, I am running 64-bit:
`$ python -c "import sys; import platform; print(f'{platform.platform()}\n{sys.version}')
Windows-10-10.0.19042-SP0
3.9.7 (default, Sep 16 2021, 16:59:28) [MSC v.1916 64 bit (AMD64)]`

mliberty1, 2022-03-10:
Ok.  The Joulescope python packages should definitely run on your computer.

I think that we have a few options:
1. Troubleshoot the existing Anaconda installation
2. Use a [virtual environment](https://docs.python.org/3/library/venv.html) with the Anaconda installation
3. Run vanilla [python](https://www.python.org/)

Options (2) and (3) should just work as they will pull in fresh dependencies.  For (3), you will need to set your path to find the "correct" python.  You can verify using `where python` and `python -VV`.  You can then `pip3 install joulescope`, and it should just work.

For option (1), I do not have much recent experience with Anaconda.  The anaconda package manager solved the binary distribution problem before python wheels, but it's much less useful today.  I suspect that pip/pypi and conda are not playing nicely to manage the dependencies correctly.  For what it's worth, we built the `joulescope` wheel with numpy 1.22.2.  It looks like the latest is now 1.22.3, and I just confirmed that 1.22.3 works with pyjoulescope 0.9.11.

In stock python, you would update using:

    pip3 install -U numpy

Not sure how you best do this in Anaconda without breaking other things.

If Anaconda still doesn't work and you want to try approach (1), could you post the output of:

    pip freeze

e-chesoni, 2022-03-10:
Hi @mliberty1 !

Thank you for the suggestions--they were very applicable in troubleshooting this issue! It seems the error was caused by incompatibility between joulescope and one (or more of) the packages I had accumulated in the base Anaconda env.

I was able to fix the error by removing the joulescope package from the base env, then creating a separate conda env, and installing the package there. As a side note, I also needed to downgrade pywin32 to 300 to import joulescope.

Thank you for the help!

closed 2022-03-10 by e-chesoni

mliberty1, 2022-03-11:
Great to hear that you are up and working!

It is strange that you had to downgrade pywin32.  The Python 3.9 amd64 wheel was built with pywin32 303, which is the latest.  I will certainly keep this in mind if I hear other issues.  Thanks for the info!