jetperch/pyjoulescope #28: Reboot loop of joulescope when the UI tries to connect

korken89, 2022-08-25:
Hi,

We've encountered that our joulescope never connects to the UI and that the UI gives errors on the USB communication and resets the scope.
Before the UI start, `dmesg` shows that the joulescope is connected and has a connection. What is your recommendation?

We have installed via `pip`, and are running on Arch Linux. It did work before with an earlier version as well, I just updated to the latest release and it stoped working.

I have attached the log of the UI and dmesg.

dmesg:

```
[19572.664191] usb 1-1.2.2.1: new high-speed USB device number 11 using xhci_hcd
[19572.776938] usb 1-1.2.2.1: New USB device found, idVendor=16d0, idProduct=0e88, bcdDevice= 0.03
[19572.776947] usb 1-1.2.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[19572.776950] usb 1-1.2.2.1: Product: Joulescope
[19572.776952] usb 1-1.2.2.1: Manufacturer: Jetperch LLC
[19572.776953] usb 1-1.2.2.1: SerialNumber: 001636

----------------------- HERE WE START UI -----------------------

[19661.787475] usb 1-1.2.2.1: USB disconnect, device number 11
[19662.010830] usb 1-1.2.2.1: new high-speed USB device number 12 using xhci_hcd
[19662.123633] usb 1-1.2.2.1: New USB device found, idVendor=16d0, idProduct=0e88, bcdDevice= 0.03
[19662.123644] usb 1-1.2.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[19662.123648] usb 1-1.2.2.1: Product: Joulescope
[19662.123651] usb 1-1.2.2.1: Manufacturer: Jetperch LLC
[19662.123653] usb 1-1.2.2.1: SerialNumber: 001636
[19666.139138] usb 1-1.2.2.1: USB disconnect, device number 12
[19666.404176] usb 1-1.2.2.1: new high-speed USB device number 13 using xhci_hcd
[19666.518050] usb 1-1.2.2.1: New USB device found, idVendor=16d0, idProduct=0e88, bcdDevice= 0.03
[19666.518062] usb 1-1.2.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[19666.518066] usb 1-1.2.2.1: Product: Joulescope
[19666.518068] usb 1-1.2.2.1: Manufacturer: Jetperch LLC
[19666.518071] usb 1-1.2.2.1: SerialNumber: 001636
[19673.563807] usb 1-1.2.2.1: USB disconnect, device number 13
[19673.827816] usb 1-1.2.2.1: new high-speed USB device number 14 using xhci_hcd
[19673.936760] usb 1-1.2.2.1: New USB device found, idVendor=16d0, idProduct=0e88, bcdDevice= 0.03
[19673.936765] usb 1-1.2.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[19673.936766] usb 1-1.2.2.1: Product: Joulescope
[19673.936768] usb 1-1.2.2.1: Manufacturer: Jetperch LLC
[19673.936768] usb 1-1.2.2.1: SerialNumber: 001636
[19677.915509] usb 1-1.2.2.1: USB disconnect, device number 14
[19678.180843] usb 1-1.2.2.1: new high-speed USB device number 15 using xhci_hcd
[19678.293289] usb 1-1.2.2.1: New USB device found, idVendor=16d0, idProduct=0e88, bcdDevice= 0.03
[19678.293299] usb 1-1.2.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[19678.293303] usb 1-1.2.2.1: Product: Joulescope
[19678.293305] usb 1-1.2.2.1: Manufacturer: Jetperch LLC
[19678.293308] usb 1-1.2.2.1: SerialNumber: 001636
[19685.084188] usb 1-1.2.2.1: USB disconnect, device number 15
[19685.307514] usb 1-1.2.2.1: new high-speed USB device number 16 using xhci_hcd
[19685.416504] usb 1-1.2.2.1: New USB device found, idVendor=16d0, idProduct=0e88, bcdDevice= 0.03
[19685.416511] usb 1-1.2.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[19685.416514] usb 1-1.2.2.1: Product: Joulescope
[19685.416516] usb 1-1.2.2.1: Manufacturer: Jetperch LLC
[19685.416518] usb 1-1.2.2.1: SerialNumber: 001636
[19689.436109] usb 1-1.2.2.1: USB disconnect, device number 16
[19689.731026] usb 1-1.2.2.1: new high-speed USB device number 17 using xhci_hcd
[19689.843415] usb 1-1.2.2.1: New USB device found, idVendor=16d0, idProduct=0e88, bcdDevice= 0.03
[19689.843426] usb 1-1.2.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[19689.843430] usb 1-1.2.2.1: Product: Joulescope
[19689.843433] usb 1-1.2.2.1: Manufacturer: Jetperch LLC
[19689.843435] usb 1-1.2.2.1: SerialNumber: 001636
[19696.604369] usb 1-1.2.2.1: USB disconnect, device number 17
[19696.827498] usb 1-1.2.2.1: new high-speed USB device number 18 using xhci_hcd
[19696.936635] usb 1-1.2.2.1: New USB device found, idVendor=16d0, idProduct=0e88, bcdDevice= 0.03
[19696.936646] usb 1-1.2.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[19696.936650] usb 1-1.2.2.1: Product: Joulescope
[19696.936653] usb 1-1.2.2.1: Manufacturer: Jetperch LLC
[19696.936655] usb 1-1.2.2.1: SerialNumber: 001636
[... 558 lines cut ...]
OSError: usb.Device ControlTransferResponse(result=5)
WARNING:2022-08-25 13:34:34,726:driver.py:637:joulescope.driver:calibration_read transfer failed 1
WARNING:2022-08-25 13:34:34,878:device.py:504:joulescope.usb.libusb.device:libusb_submit_transfer [control] => -4
WARNING:2022-08-25 13:34:34,879:device.py:524:joulescope.usb.libusb.device:device_removed
ERROR:2022-08-25 13:34:34,879:main.py:918:joulescope_ui.main:during parameter_set
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/joulescope_ui/main.py", line 916, in _on_device_parameter
    self._device.parameter_set(topic, value)
  File "/usr/lib/python3.10/site-packages/joulescope/driver.py", line 443, in parameter_set
    self._stream_settings_send()
  File "/usr/lib/python3.10/site-packages/joulescope/driver.py", line 722, in _stream_settings_send
    _ioerror_on_bad_result(rv)
  File "/usr/lib/python3.10/site-packages/joulescope/driver.py", line 48, in _ioerror_on_bad_result
    raise IOError('usb.Device %s' % (rv,))
OSError: usb.Device ControlTransferResponse(result=5)
ERROR:2022-08-25 13:34:51,600:driver.py:1338:joulescope.driver:while scanning for devices
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/joulescope/driver.py", line 1331, in scan
    devices = usb.scan(name)
  File "/usr/lib/python3.10/site-packages/joulescope/usb/libusb/device.py", line 1028, in scan
    serial_number = _get_string_descriptor(dh, descriptor.iSerialNumber)
  File "/usr/lib/python3.10/site-packages/joulescope/usb/libusb/device.py", line 351, in _get_string_descriptor
    raise RuntimeError('control transfer could not get string descriptor: %d' % (rv, ))
RuntimeError: control transfer could not get string descriptor: -1
```

mliberty1, 2022-08-25:
HI @korken89 - Sorry to hear that you are having issues with your Joulescope, but thank you for the logs!  I took a look, and I suspect that the sensor-side power supply is not powering up correctly on your Joulescope unit.  I have a few things to try:

1. Can you plug your Joulescope directly into a USB port on your host computer?  Please avoid any hubs or docks.  This failure mode is possible if a hub/dock is unable to supply enough power to your Joulescope unit.  Any change?

2. If you see no change with (1), do you have access to a Microsoft Windows computer?  Can you try using this Joulescope unit with that computer?  This will help isolate the issue to the Joulescope unit or the Arch computer.

korken89, 2022-08-25:
Hi, thanks for the input!

We gave it a try on an Ubuntu computer, and it works fine. So this got us testing a bit.
After some experimentation we noticed that it only wanted to work if we connected it to a powered USB Hub (external power).
My guess is that the ports on the laptop can't provide the current for some reason, while the Hub can. 
It's a Thinkpad E14 laptop that we are using, so I guess the budget laptops might not be the best...

So all in all, it is working now! :) Thanks again for your time!

mliberty1, 2022-08-25:
Hi @korken89 - That's a first!  I have never heard of a computer (laptop or desktop) not being able to supply sufficient power and still enumerating the device.  The USB 2 spec allows devices to request up to 500 mA which is what Joulescopes request.  USB battery charging and USB-PD allows devices to request a lot more current.  

I wonder if that laptop is very aggressive in limiting current.  Joulescopes do draw the full amount when the sensor-side powers up.  Joulescopes tend to draw about 250 mA in normal operation.

Either way, great to hear that you are back up and working!

closed 2022-08-25 by mliberty1