jetperch/pyjoulescope #29: Support for keeping Joulescope turned on without having to have UI open

Decee1, 2022-09-27:
Hello,
I got a request which have annoyed me A LOT
I got a test setup where i need to do system test of a DUT. 
And here i use Joulescope for measuring current. This turned out to be a big regret as it can enable/disable output. 
And i cannot find any way of doing this using the Python library. What i want is to force it to be turned on until i turn it off again. 

Currently my only solution is to keep the UI running but then i can't remote control the Joulescope which makes it useless.. 
Since the PyJoulescope_ui can do it. I would estimate there is API support for it somehow

mliberty1, 2022-09-27:
Hi @Decee1 - I am not entirely sure I understand what you are trying to do.  If you just want to turn power on/off to the DUT, check out the [dut_power](https://github.com/jetperch/pyjoulescope_examples/blob/main/bin/dut_power.py) example.  Does this do what you want?

Decee1, 2022-09-27:
Hi @mliberty1 
That was exactly what i needed.
It was just not documented in the API. Not a big fan of the 'parameter_set' function. Seems like a lazy solution to just making proper API documentation for each of the function calls.
But hey it seems like it works!

Does this time out after some time or will the output stay 'on' until for example the USB port no longer is busy?

mliberty1, 2022-09-27:
The JS110 will remain in the same state until you run another script that changes the state or the host computer no longer provides power to the JS110, such as when the host computer powers off or goes to sleep.

mliberty1, 2022-09-27:
You can find a full list of JS110 parameters and their descriptions in [parameters_v1.py](https://github.com/jetperch/pyjoulescope/blob/6b92e38c7ede9a0514d03d19dbb67c8d6b2490be/joulescope/parameters_v1.py).

Since you have a working solution, I am going to close this issue.  If you run into more issues, feel free to reopen this issue or create a new issue.

closed 2022-09-27 by mliberty1