jetperch/pyjoulescope #33: Opening the device causes brief interruption of power to DUT

andreasmueller, 2023-04-19:
Hi

When I open the Joulescope (JS110), the power to the DUT is briefly interrupted:

```python
import joulescope
js = joulescope.scan_require_one(config='auto')
js.open()  # <-- power is toggled here
```
This seems to be a regression of #11 ( same behavior), but likely not related. Version 0.9.11 works fine for me.

mliberty1, 2023-04-19:
Thanks for reporting this issue @andreasmueller.  I agree that the JS110 should not disconnect power to the DUT with these commands.  I quickly inspected the code, and I see why this is happening.  I have not yet identified an appropriate fix.  However, you can continue using version 0.9.11, as you indicated, as long as you do not need JS220 support.

I have duplicated this behavior with my setup.  I connected the JS220 Evaluation Kit 1 to the JS110's inputs.  I also connected an oscilloscope across the JS110's OUT terminals (Current - to Voltage -), like this:
[image]

Using Thonny, I configured the JS220 EVK1:

```
evk.stop()
evk.resistance = 10
evk.power_ldo(1.0)
```

To duplicate the issue, you first need to connect the JS110 with something that sets the current range to "auto".  I used the Joulescope UI.  You can then connect again.  If the oscilloscope is set to trigger on voltage drop, you will see a short time where the voltage drops, like this:

[image]

mliberty1, 2023-04-19:
The joulecope_driver commit [ec2b4a9](https://github.com/jetperch/joulescope_driver/commit/ec2b4a95a835f60d35a76cc56eab27d10992cb2e) addresses this issue.  The fix is in pyjoulescope_driver 1.3.3, which is now available.  Here is how to upgrade:

    python -m pip install U pyjoulescope_driver

Replace "python" with however you normally launch python, such as python3 or the full path.

I am going to close this issue, but please reopen if you still see still see the problem after updating!

closed 2023-04-19 by mliberty1

andreasmueller, 2023-04-20:
I just tested it and can confirm pyjoulescope-driver 1.3.3 solves the issue.

Thank you for the quick reaction & very quick fix!