jetperch/pyjoulescope #34: Joulescope on raspberrpi

foofel, 2023-10-18:
We are trying to create a setup for long term data collection using the joulescope and timescaledb. When trying to run the joulescope example data streamer on the raspberry one core is always at 100%, is this normal or does this mean the raspberry can not keep up with the data collected? (We are want to collect at 1000Hz). As far as i understand all data is streamed to the host computer at 1Msps and then reduced on the host? It looks like all data is arriving as expected but would i notice/ how do i notice if we had data drops?

foofel, 2023-10-18:
Hi @mliberty1, 

yes it is raspberrypi4 with the default 64 bit raspbian and a Joulescope JS220. I took the example from the [read_by_callback.py](https://github.com/jetperch/pyjoulescope_examples/blob/main/bin/read_by_callback.py). I haven't really looked at the server yet. Using statistics at 1Khz sounds promising, is there also an example for that?

mliberty1, 2023-10-18:
The easiest way to get statistics is by running:

    python -m pyjoulescope_driver statistics

You can provide the `--frequency 1000` optional argument to get 1 kHz data.  This command displays the statistics to stdout using the newer pyjoulescope_driver directly.  You can also do the same from a python script.  You can find the code for this pyjoulescope_driver entry point [here](https://github.com/jetperch/joulescope_driver/blob/main/pyjoulescope_driver/entry_points/statistics.py).

You should get NaN (not a number) values if samples are being dropped when streaming full-rate samples or downsample samples.

foofel, 2023-10-18:
Thanks!

I will look into the statistics and getting NaNs is easily detectable :)

closed 2024-02-14 by mliberty1

foofel, 2024-02-24:
Just as short feedback, the statistics work well, thanks :)