jetperch/pyjoulescope #35: Regular expression problem in 'units.py'

dbwalker0min, 2024-02-07:
On line 27 of unnits.py, there is a line:
```python
RE_IS_NUMBER = re.compile('^\s*([-+]?[0-9]*\.?[0-9]+)\s*(.*)')
``` 
In my version of PyCharm (and during run time running Python 3.11), it complains:
```
<input>:1: SyntaxWarning: invalid escape sequence '\s'
``` 
This seems to be true; it is an escape sequence in a string that isn't valid. I think it should be:
```python
RE_IS_NUMBER = re.compile(r'^\s*([-+]?[0-9]*\.?[0-9]+)\s*(.*)')
```

dbwalker0min, 2024-02-07:
It is innocuous. Python interprets it properly, though loudly.

```
print('\s')
<input>:1: SyntaxWarning: invalid escape sequence '\s'
<input>:1: SyntaxWarning: invalid escape sequence '\s'
\s
```

mliberty1, 2024-02-07:
Interestingly, the unit test runs without errors or error messages for me:

    python -m unittest joulescope.test.test_units
    
Adding the "r" string modifier gives the same results.  As you noted, it is correct to have either "\\" or the "r" string modifier.

And here is my python version:

    python -VV
    Python 3.11.7 (tags/v3.11.7:fa7a6f2, Dec  4 2023, 19:24:49) [MSC v.1937 64 bit (AMD64)]

We will add the "r" string modifier for correctness.

mliberty1, 2024-02-14:
Fixed in 1.1.12

closed 2024-02-14 by mliberty1