jetperch/pyjoulescope #36: Python32-bit is not installed

gavin100200, 2024-02-28:
An error occurs when installing joulescope in python3.10.1 32-bit
ERROR: Failed building wheel for pyjoulescope-driver
ERROR: Could not build wheels for pyjoulescope-driver, which is required to install pyproject.toml-based projects

Is it not possible to install 32-bit only 64-bit？

mliberty1, 2024-02-28:
Hi @gavin100200 - We only officially support 64-bit python on 64-bit operating systems.  We only distribute pre-compiled wheels for 64-bit platforms.  However, you should be able to run 32-bit python on 32-bit operating system.

The error message you posted is definitely truncated.  The actual error would have been displayed previously.  Since we do not distribute 32-bit wheels, this machine has to be configured to build source code, which I suspect is the issue.

Some questions:
1. What operating system are you using?
2. What is the output of `python -VV` (two V)?  You may need to run `python3 -VV` on linux.

mliberty1, 2024-02-28:
See https://forum.joulescope.com/t/python3-10-1-32-bit-is-not-installed/763

mliberty1, 2024-11-08:
Will not fix since all major operating systems are now 64-bit.

closed 2024-11-08 by mliberty1