jetperch/pyjoulescope #37: wrong start time in JLS

WeiminWangKolmostar, 2024-04-16:
I use both the https://github.com/jetperch/pyjoulescope_examples/blob/main/bin/capture_jls_v2.py and https://github.com/jetperch/pyjoulescope_examples/blob/main/bin/capture_jls_v1.py scripts to capture samples to a JLS v1 & v2 file, when I load the JLS files in joulescope GUI client, it always shows the sample start time is 2018-01-01.

[image]

mliberty1, 2024-04-16:
Hi @WeiminWangKolmostar and thanks for creating this issue.  The UTC time is a newer feature that never made it into the `pyjoulescope` binding.  If you want this feature today, I recommend using:

```
python3 -m pyjoulescope_driver out.jls
```

You can provide the `--help` argument for more details about the available arguments.

Regarding the reported issue:
1. JLS v1 does not support UTC time and never will.  JLS v1 is also deprecated.
2. JLS v2 does support UTC time, but [`JlsWriter`](https://github.com/jetperch/pyjoulescope/blob/main/joulescope/jls_v2_writer.py) in pyjoulescope does not.  We will keep this item open.
3. For the time being, you can use pyjoulescope_driver instead.  Both the [record entry point](https://github.com/jetperch/joulescope_driver/blob/main/pyjoulescope_driver/entry_points/record.py) and [record module](https://github.com/jetperch/joulescope_driver/blob/main/pyjoulescope_driver/record.py) add UTC time correctly.

mliberty1, 2026-10-09:
Fixed in 1.7.0

closed 2026-10-09 by mliberty1