jetperch/pyjoulescope #38: Install failes on Py312 (and Win11)

orgua, 2024-06-05:
Problem is that this packages wants an outdated version of `pymonocypher` that has to be compiled for py3.12. Newer versions of that package already seem to be available as wheel. Could you please fix that?

While trying to make compilation work I installed the MS C++ build tools v140 through v143 (7 GB). Also restarted console and computer, but the error persisted:

```Shell
[pipenv.exceptions.InstallError]:   × Building wheel for pymonocypher (pyproject.toml) did not run successfully.
[pipenv.exceptions.InstallError]:   │ exit code: 1
[pipenv.exceptions.InstallError]:   ╰─> [5 lines of output]
[pipenv.exceptions.InstallError]:       running bdist_wheel
[pipenv.exceptions.InstallError]:       running build
[pipenv.exceptions.InstallError]:       running build_ext
[pipenv.exceptions.InstallError]:       building 'monocypher' extension
[pipenv.exceptions.InstallError]:       error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
[pipenv.exceptions.InstallError]:       [end of output]
```

I kept on installing more from the MS tooling and suddenly it started compiling but failed because of some missing header-files.

mliberty1, 2024-06-05:
Unfortunately, Monocypher 4 is a major release and contains changes that are not backwards compatible with Monocypher 3.  We are stuck with Monocypher and pymonocypher 3.  However, we also maintain pymonocypher.

I just published [pymonocypher 3.1.3.3](https://pypi.org/project/pymonocypher/3.1.3.3/) that includes Python 3.12 precompiled wheels.  Try `pip install -U joulescope` again.

Does this fix the problem for you?

orgua, 2024-06-05:
thanks for the quick fix! To get some measurements I downgraded to py311 for now - but tests on a second device showed success!

closed 2024-06-05 by orgua