jetperch/pyjoulescope #39: How to enumerate a list of valid topics for publish()

orgua, 2024-06-05:
Coming from the UI it seems possible to:

- disable the voltage measurement
- change the sample frequency
- set a specific range

Looking through the example-repo I found a solution for the last task:

```python
js.publish("s/i/range/mode", "manual")
js.publish("s/i/range/select", range_value)
```

Looking through the users-guide.pdf I found valid ranges:

```python
ranges = ["180 mA", "18 mA", "1.8 mA", "180 µA", "18 µA"]
```

Funny enough, that I can't just copy values from the table on page 14/49 when it comes to micro-ampere, but also the micro-symbol from my keyboard got rejected with:

```
RuntimeError: jsdrv_publish failed 5 PARAMETER_INVALID | The parameter value is invalid | u/js220/002649/s/i/range/select
```

After some discouragement the third version of the micro-symbol worked though.

Beside that minor hurdle my main questions are:

- are these settings (disabling Voltage and changing sample rate) possible? the user guide generalizes 2 MSPS, so I have hope.
- is there a documented list of topics or is it possible to generate one? Keithley SMUs spill out the whole settings-tree on request. With the joulescope I have trouble finding the topic names at all.

mliberty1, 2024-06-05:
We have two different API levels.  For the `joulescope` package that I think you are using, you can find the parameters here:
https://github.com/jetperch/pyjoulescope/blob/main/joulescope/parameters_v1.py

Starting with the JS220, we implemented a LOT more on the instrument, and the instrument is responsible for the available parameters.  The `joulescope` package wraps the device topics.  You can list the device topics using the `pyjoulescope_driver` entry point, like this:

```
python -m pyjoulescope_driver info -v "*"
```

mliberty1, 2024-06-05:
Yes, the "micro" symbol is slightly cursed in Unicode.  I keep this handy text file handy:

```
µ = \u00B5 micro sign (not μ = \u03BC greek small letter mu)
Δ = \u0394 greek capital letter delta
Ω = \u03A9 greek capital letter omega
± = \u00b1 plus-minus sign
τ

→ ↔ ←
⭐
π∫σωτ°√
✓✔❌
©®™
⚠
①②③④⑤⑥⑦⑧⑨⑩

https://unicodearrows.com/all-arrows
```

orgua, 2024-06-05:
Okay - so it is possible on driver-level. That helps a lot - Thanks again!

orgua, 2024-06-05:
Sorry for opening that up again, but I am not able to implement the desired functionality.

## Reducing Sample Frequency

```python
with joulescope.scan_require_one(config='auto') as js:
   js.publish("/h/fs", "1 kHz")
   print(js.output_sampling_frequency)
   print(js.sampling_frequency)
   data = js.read(contiguous_duration=duration)
```

both prints() still report `1000000` and read() never exits after changing frequency.

## Allow 2 MSPS Current Measurement

The user guide implies 2 MSPS and I'd like to switch off voltage-measurements and only measure current with full rate. Even though that's above the lowpass cutoff it would be nice to only get the requested data. While `/s/i/range/mode` & `/s/i/range/select` has an `off`-option, the `/s/v/range` has not. Setting '/s/v/ctrl'=0 still produces that measurement stream.

There is now way, right?

mliberty1, 2024-06-05:
So, if you are using the `joulescope` package, it's better to use the `joulescope` parameters as defined here:
https://github.com/jetperch/pyjoulescope/blob/main/joulescope/parameters_v1.py

So, you would just `js.parameter_set('sampling_frequency', 1000)`

We have [lots of examples](https://github.com/jetperch/pyjoulescope_examples/tree/main/bin).

Now, the `joulescope` v1 backend uses `pyjoulescope_driver`, and it does implement the `publish` method.  However, I think the problem is the leading '/'.  You just need `js.publish("h/fs", "1 kHz")`.

It sounds like you are trying to accomplish some task or measurement.  What is the actual objective?  There is a good chance we have an example or even a command-line tool that will make this much faster.

mliberty1, 2024-08-21:
Closed assuming that the previous response solved the problem.  If not, please comment and reopen!

closed 2024-08-21 by mliberty1