jetperch/pyjoulescope #41: Add arm linux wheel deploy and build

daltonv, 2024-06-28:
It would be great if an arm linux wheel would be prebuilt, so arm users don't have to build the wheel locally.

Should be possible to use qemu to build the wheel on github actions cloud runners as shown at -> https://github.com/pypa/cibuildwheel/blob/main/examples/github-with-qemu.yml

mliberty1, 2024-06-28:
I really do not want to support a qemu build across all of our python packages.  Maintaining the build system is already enough work.  GitHub is working to support aarch64.  See [this issue](https://github.com/actions/runner-images/issues/5631).  It looks like GitHub [announced](https://github.blog/2024-06-03-arm64-on-github-actions-powering-faster-more-efficient-build-systems/) a Public Beta earlier this month.  From that same article:

_We expect to begin offering Arm runners for open source projects by the end of the year._

Once GitHub adds a aarch64 runner, it should be easy to add support.  So, we may have better support by the end of the year.

daltonv, 2024-06-28:
Fair enough. I didn't know about the aarch64 runner from GitHub. I can wait until then. It will make my docker builds easier anyway.