jetperch/pyjoulescope #42: Error in SampleBuffer when changing buffer size often

EmilMelgaard, 2024-08-09:
Sometimes I get a ValueError when I change the buffer duration to match the sampling duration just before reading.

I'm using python 3.11 64-bit, joulescope version 1.1.15, pyjoulescope-driver version 1.5.4 and a Joulescope JS220.

Example of the error:

```python
[joulescope_log.txt](https://github.com/user-attachments/files/16560628/joulescope_log.txt)

Traceback (most recent call last):
  File "C:\Users\AU000B9D\AppData\Local\Programs\Python\Python311\Lib\site-packages\IPython\core\interactiveshell.py", line 3508, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-4-aebd176e0f8e>", line 12, in <module>
    device.read(contiguous_duration=t)
  File "C:\Users\AU000B9D\AppData\Local\Programs\Python\Python311\Lib\site-packages\joulescope\v1\device.py", line 564, in read
    data = self.stream_buffer.samples_get(start_id, end_id, fields=['current', 'voltage'])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\AU000B9D\AppData\Local\Programs\Python\Python311\Lib\site-packages\joulescope\v1\stream_buffer.py", line 387, in samples_get
    out = b.get_range(start, stop)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\AU000B9D\AppData\Local\Programs\Python\Python311\Lib\site-packages\joulescope\v1\sample_buffer.py", line 171, in get_range
    raise ValueError(f'get {start}:{end} out of available range {s_start}:{s_end}')
ValueError: get 4080923057:4084907534 out of available range 4080923789:4084923788
```

Example code to reproduce the error:

```python
import joulescope

device = joulescope.scan_require_one()
device.open()

i = 0
while True:
    for t in (2, 4):
        i += 1
        print(f"Measure {i} (t = {t})")
        device.buffer_duration = t
        device.read(contiguous_duration=t)
```

I don't get the error the same time every run, but I have attached a log of a full run with info logging enabled.

EmilMelgaard, 2024-08-09:
Full log: [joulescope_log.txt](https://github.com/user-attachments/files/16560638/joulescope_log.txt)

mliberty1, 2024-08-09:
Hi @EmilMelgaard - When used this way, the sample buffer needs to be large enough for the read and at least one sample chunk.  An extra second should be more than enough.  Does changing this line fix the issue?

```
device.buffer_duration = t + 1
```

If you are making measurements on the order of seconds, there is also no need to resize the buffer.  The default 30 second buffer should work just fine.

EmilMelgaard, 2024-08-09:
Hi @mliberty1, Thank you, adding a second seems to fix the issue as far as I can see. In my real code I measure up to four minutes, I just reduced the time in the example reduce the time to reproduce :-)

mliberty1, 2024-08-09:
Great!  Note that the buffer requires about 15 MB/s, and then provides a copy which doubles the RAM requirement.  I would not recommend going much beyond 4 minutes duration using this method.  For longer captures, you can save directly to your own buffer to avoid the copy.  See [read_by_callback.py](https://github.com/jetperch/pyjoulescope_examples/blob/main/bin/read_by_callback.py) and [read_by_pyjoulescope_driver.py](https://github.com/jetperch/pyjoulescope_examples/blob/main/bin/read_by_pyjoulescope_driver.py).  For even longer captures, just record to JLS first.

I will close this issue, but feel free to reopen if you still have concerns!

closed 2024-08-09 by mliberty1