jetperch/pyjoulescope #43: upgrade to numpy >2.0

9y2070m, 2024-09-17:
Hi,

I just tried to implement the joulescope into my CI/CD environment.

Unfortunately, I have a dependency mismatch with `numpy`.
All the other libraries used are already on `numpy`=^2.0.0 only the joulscope is limiting it to <2.

Do you know if there is anything to do about this?
When will the change be done?
Can I offer support on that?

BR
Thomas

mliberty1, 2024-09-17:
We do plan on migrating to Numpy 2.x, but we have no schedule yet.  Numpy just released 2.0.0 in June, and is already at 2.1.1.  My thought is to let things stabilize for another few months before migrating Joulescope software.

To complicate things, our projects use Cython and Nuitka to compile Python.  These tools also need to support Numpy 2.  When I tried back in June, they still had challenges, but I am sure they are making progress.

To migrate Joulescope software to Numpy 2.x, we need to update the following projects:

* [jls](https://github.com/jetperch/jls) (pyjls)
* [joulescope_driver](https://github.com/jetperch/joulescope_driver) (pyjoulescope_driver)
* [pymonocypher](https://github.com/jetperch/pymonocypher]
* [pyjoulescope](https://github.com/jetperch/pyjoulescope)
* [pyjoulescope_ui](https://github.com/jetperch/pyjoulescope_ui)

We also need to upgrade some private packages.  Overall, this likely ends up being a significant amount of work that we need to schedule.

My recommendation for now is to continue to use Numpy 1.x and other older package versions that continue to support Numpy 1.x. 

## References

* [Numpy 2 migration guide](https://numpy.org/devdocs/numpy_2_0_migration_guide.html)
* [Cython](https://cython.org/)
* [Nuitka](https://nuitka.net/)

9y2070m, 2024-10-01:
👍 I'll try to use an older version.

Please feel free to reach out if you need support for upgrading.

mliberty1, 2024-11-08:
Completed in 1.2.0

closed 2024-11-08 by mliberty1