jetperch/pyjoulescope #44: RuntimeError  when parameter_set

byteptr, 2024-09-18:
I have a RuntimeError when trying to ser parameter sampling_frequency to 1Khz, If I comment the line of the parameter change the error dissapears.

"device_open(u/js110/001306) could not open device path \\?\usb#vid_16d0&pid_0e88#001306#{576d606f-f3de-4e4e-8a87-065b9fd21eb0}
open_ll failed"

Steps to reproduce: 

****************************************
import joulescope  
import numpy as np  
js = joulescope.scan_require_one(config='auto')  
js.parameter_set('sampling_frequency',1000)  # If I add this line, js cannot read data
with js:  
    data = js.read(contiguous_duration=0.2)  
****************************************

joulescope version 1.1.13
Python 3.9.13
OS WIndows 10
Note, I'm running under Jupyter Notebook
Can anyone guide me on this, please ? 
Many Thanks In advance.

mliberty1, 2024-11-08:
Could not duplicate and no response.

@byteptr - Please reopen if this is still an issue!

closed 2024-11-08 by mliberty1

byteptr, 2024-11-09:
Hi, 
sorry for the delays. I came back into the script the last week, since we re-opened the project.
The I spent some time debugging, and I found a bug from our side. It seems the joulescope instance was already running on a "spurious thread", I removed It, then all works fine.