jetperch/pyjoulescope #46: Changing the `output_sampling_frequency` causes invalid results.

mentalAdventurer, 2025-01-28:
Changing the `output_sampling_frequency` causes invalid results for me.

**Platform**: Linux
**Python**: `3.10.15`

## This works fine

```py
js = joulescope.scan_require_one(config="auto")
js.open()
js.start()
time.sleep(2)
js.stop()
start_id, end_id = js.stream_buffer.sample_id_range
print(f"{start_id=}, {end_id=}")
print(f"diff: {end_id-start_id}")
```

**Output**:
```sh
start_id=1489056944, end_id=1491022544
diff: 1965600
```

## This does not

```py
js = joulescope.scan_require_one(config="auto")
js.output_sampling_frequency = int(js.output_sampling_frequency/2)
js.open()
js.start()
time.sleep(2)
js.stop()
start_id, end_id = js.stream_buffer.sample_id_range
print(f"{start_id=}, {end_id=}")
print(f"diff: {end_id-start_id}")
```

Output:
```sh
start_id=1529723080, end_id=1529723080
diff: 0
```

I hope I'm right in assuming that the `output_sampling_frequency` can be changed?  🙈

mentalAdventurer, 2025-01-28:
Yes, the updated version of the script does work on Linux! 
I agree with the removal of the setter. It caused some confusion for me, exspecifically because there is some logic behind it that makes it seem functional 😅. 

Thank you for your help 💓

mliberty1, 2025-01-28:
Removed `output_sampling_frequency` setter in [1.2.2](https://github.com/jetperch/pyjoulescope/releases/tag/v1.2.2).

closed 2025-01-28 by mliberty1